fhunleth/elixir_ale #1: Error at installation on beagleboard black

ChristopheBelpaire, 2015-03-09:
Hello,
I tried to install the library on a beaglebone black, with Elixir 1.0.3 and OTP 17.4, both compiled from source.
I have this error when I try to install
```bash
mix compile
warning: the VM is running with native name encoding of latin1 which may cause Elixir to malfunction as it expects utf8. Please ensure your locale is set to UTF-8 (which can be verified by running "locale" in your shell)
cc src/ale_main.o src/gpio_port.o src/i2c_port.o src/spi_port.o src/erlcmd.o -L/usr/lib/erlang/usr/lib -lei  -o priv/ale
/usr/bin/ld: cannot find -lei
collect2: ld returned 1 exit status
make: *** [priv/ale] Error 1
** (MatchError) no match of right hand side value: 2
    mix.exs:4: Mix.Tasks.Compile.ElixirAle.run/1
    (mix) lib/mix/tasks/compile.ex:59: anonymous fn/2 in Mix.Tasks.Compile.run/1
    (elixir) lib/enum.ex:977: anonymous fn/3 in Enum.map/2
    (elixir) lib/enum.ex:1261: Enum."-reduce/3-lists^foldl/2-0-"/3
    (elixir) lib/enum.ex:977: Enum.map/2
    (mix) lib/mix/tasks/compile.ex:58: Mix.Tasks.Compile.run/1
    (mix) lib/mix/cli.ex:55: Mix.CLI.run_task/2
```

Any idea?
Thanks!

fhunleth, 2015-03-09:
Are you running Debian on the BBB? I almost never run it, so I think that I don't have the path to libei.a set correctly. Could you check if you have libei.a in /usr/lib/erlang/lib/erl_interface-3.7.20/lib or in /usr/lib/erlang/usr/lib? My assumption was the latter. If you do have it in the erl_interface directory, could you try adding that directory to the linker options in the Makefile. E.g., -L/usr/lib/erlang/lib/erl_interface-3.7.20/lib? If you don't have the library at all, then you probably need to run apt-get to get erlang-dev or something similar. Please let me know what you find so that I can update the Makefile and/or docs?

TattdCodeMonkey, 2015-03-09:
@ChristopheBelpaire I had a similar issue when setting up erlang & elixir on the Raspberry Pi 2. I set the default locale to en_UTF8 and that corrected the issue for me.

ChristopheBelpaire, 2015-03-10:
Thanks for the fast answer!
I use debian, and indeed I to change ELR_EI_LIBDIR to :
```ERL_EI_LIBDIR ?= /usr/local/lib/erlang/usr/lib```
now let's try this library :+1:

fhunleth, 2015-03-12:
@ChristopheBelpaire - I just updated the Makefile to search for libei.a in /usr/local/lib/erlang in fae035b82f9a22c8af674adb9847fd7516c98e35. Could you `git pull` and see if it builds out of the box for you?

ChristopheBelpaire, 2015-03-13:
@fhunleth I tried with updated version and it compiled without problem :+1: 
I also tried the I2C functions and it seems works fine.
Thanks again ;)
```bash
christophe@test11:~/ea/elixir_ale$ mix compile
warning: the VM is running with native name encoding of latin1 which may cause Elixir to malfunction as it expects utf8. Please ensure your locale is set to UTF-8 (which can be verified by running "locale" in your shell)
cc -c -I/usr/local/lib/erlang/lib/erl_interface-3.7.20/include -O2 -Wall -Wextra -Wno-unused-parameter -o src/ale_main.o src/ale_main.c
cc -c -I/usr/local/lib/erlang/lib/erl_interface-3.7.20/include -O2 -Wall -Wextra -Wno-unused-parameter -o src/gpio_port.o src/gpio_port.c
cc -c -I/usr/local/lib/erlang/lib/erl_interface-3.7.20/include -O2 -Wall -Wextra -Wno-unused-parameter -o src/i2c_port.o src/i2c_port.c
src/i2c_port.c: In function 'i2c_write':
src/i2c_port.c:77:35: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
src/i2c_port.c: In function 'i2c_read':
src/i2c_port.c:95:34: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
cc -c -I/usr/local/lib/erlang/lib/erl_interface-3.7.20/include -O2 -Wall -Wextra -Wno-unused-parameter -o src/spi_port.o src/spi_port.c
src/spi_port.c: In function 'spi_transfer':
src/spi_port.c:110:19: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
src/spi_port.c:111:19: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
cc -c -I/usr/local/lib/erlang/lib/erl_interface-3.7.20/include -O2 -Wall -Wextra -Wno-unused-parameter -o src/erlcmd.o src/erlcmd.c
cc src/ale_main.o src/gpio_port.o src/i2c_port.o src/spi_port.o src/erlcmd.o -L/usr/local/lib/erlang/lib/erl_interface-3.7.20/lib -lei  -o priv/ale
Compiled lib/i2c.ex
Compiled lib/gpio.ex
Compiled lib/spi.ex
Generated elixir_ale.app
```

closed 2015-03-13 by ChristopheBelpaire

infiniteNIL, 2015-05-16:
Hey, it looks like I'm having this same problem on a Raspberry Pi. Any suggestions?

infiniteNIL, 2015-05-16:
Actually, it's a different problem. I'll open up a different issue.