fhunleth/elixir_ale #2: Compile Error on Raspberry Pi

infiniteNIL, 2015-05-16:
When I do mix compile on the Raspberry Pi, I get this:

cc -c -I/usr/lib/erlang/usr/include -O2 -Wall -Wextra -Wno-unused-parameter -o src/ale_main.o src/ale_main.c
cc -c -I/usr/lib/erlang/usr/include -O2 -Wall -Wextra -Wno-unused-parameter -o src/gpio_port.o src/gpio_port.c
cc -c -I/usr/lib/erlang/usr/include -O2 -Wall -Wextra -Wno-unused-parameter -o src/i2c_port.o src/i2c_port.c
In file included from src/i2c_port.c:34:0:
/usr/include/linux/i2c-dev.h:38:8: error: redefinition of ‘struct i2c_msg’
/usr/include/linux/i2c.h:67:8: note: originally defined here
/usr/include/linux/i2c-dev.h:90:7: error: redefinition of ‘union i2c_smbus_data’
/usr/include/linux/i2c.h:125:7: note: originally defined here
src/i2c_port.c: In function ‘i2c_transfer’:
src/i2c_port.c:94:17: warning: pointer targets in assignment differ in signedness [-Wpointer-sign]
src/i2c_port.c:99:17: warning: pointer targets in assignment differ in signedness [-Wpointer-sign]
make: *** [src/i2c_port.o] Error 1
** (MatchError) no match of right hand side value: 2
    mix.exs:4: Mix.Tasks.Compile.ElixirAle.run/1
    (elixir) lib/enum.ex:1043: anonymous fn/3 in Enum.map/2
    (elixir) lib/enum.ex:1385: Enum."-reduce/3-lists^foldl/2-0-"/3
    (elixir) lib/enum.ex:1043: Enum.map/2
    (mix) lib/mix/tasks/compile.all.ex:15: Mix.Tasks.Compile.All.run/1
    (mix) lib/mix/tasks/compile.ex:64: Mix.Tasks.Compile.run/1
    (mix) lib/mix/cli.ex:60: Mix.CLI.run_task/2

fhunleth, 2015-05-16:
That's not good. Are you using Raspian?

infiniteNIL, 2015-05-16:
Yes, I believe so. I got it to work by commenting out the include of i2c-dev.h. At least it compiled. I haven't gotten far enough to verify the functionality.

TattdCodeMonkey, 2015-07-28:
@infiniteNIL is i2c enabled on your pi? 

I compile this library on my pi with Raspian frequently without issues.

infiniteNIL, 2015-07-29:
I’m using i2c in my project so does that mean its enabled? I don’t see how it would compile at all without it.

______________________
Rod Schmidt
infiniteNIL Software — Mac OS X & iPhone Software
Custom development for Mac OS X, iPhone, and the web
infinitenil.com <http://www.infinitenil.com/>

TattdCodeMonkey, 2015-07-29:
If its working then yes. On Raspian i2c and SPI are disabled by default.
https://learn.adafruit.com/adafruits-raspberry-pi-lesson-4-gpio-setup/configuring-i2c

fhunleth, 2015-11-24:
Closing this issue, since it suspect that it is resolved now. I added a link the Adafruit instructions for setting up SPI and I2C in the README, since I think that will help many people using the library.

closed 2015-11-24 by fhunleth