fhunleth/elixir_ale #3: Can't load library when elixir_ale is included as dependency in a project

ChristopheBelpaire, 2015-06-05:
Hello,
I'm trying to include the library in my project with :
```  
defp deps do
    [{:'elixir-ale', git: "git@github.com:fhunleth/elixir_ale.git", app: false}]
end
```
and when I try with iex -S mix :
```
Interactive Elixir (1.0.4) - press Ctrl+C to exit (type h() ENTER for help)
iex(1)> {:ok, pid} = I2c.start_link("i2c-1", 0x20)
** (EXIT from #PID<0.101.0>) an exception was raised:
    ** (ArgumentError) argument error
        :erlang.++({:error, :bad_name}, '/ale')
        lib/i2c.ex:59: I2c.init/1
        (stdlib) gen_server.erl:306: :gen_server.init_it/6
        (stdlib) proc_lib.erl:237: :proc_lib.init_p_do_apply/3
```
I have no problem if I do the same in the library folder (as in the readme)

Can you help ?
Thanks !

fhunleth, 2015-06-05:
That means that `:code.priv_dir(:elixir_ale)` isn't working. Could you try changing the `elixir-ale` (hyphen) to `elixir_ale` (underscore) in your deps?

ChristopheBelpaire, 2015-06-05:
Problem solved :+1: 
Thanks a lot!

fhunleth, 2015-06-05:
Great!

closed 2015-06-05 by fhunleth