fhunleth/elixir_ale #6: priv directory not copied over when used as dep

matthewphilyaw, 2015-11-23:
I'm using this project via mix in a project of mine [here](https://github.com/matthewphilyaw/motor_hat) and it works great btw, have used Erlang Ale and excited to see this. However one thing I noted is that the port that gets placed in priv/ after it's compiled doesn't get copied to the _build directory when used as a dependency.

To be clear it does get generated and that works fine, it's that my app expects it in the copied location in the _build directory.

After running mix the ale port is placed here `deps/elixir_ale/priv` but my app fails when it calls I2c.start_link because it can't find the port which it expects to be here `_build/dev/lib/elixir_ale/priv/`

Manually copying it over works fine, once where it expects no issues.

fhunleth, 2015-11-23:
Interesting. There should be a soft link to the `ale` port in the `_build` directory. Strangely enough, I could swear that I've seen this happen recently, but when I went to look into it, the file showed up.

My guess is that since I haven't updated `elixir_ale` in such a long time, I missed out on some change or fix in `mix` or something else, and now I'm doing something incorrectly in the build. Do you have a clue? If not, I'll look into it the next few days and see if I can figure out what's happening.

matthewphilyaw, 2015-11-23:
I don't have it in front of me but I never confirmed it was in the `deps/elxir_ale/_build` actually and it would make sense for it to be missing there, thus not being copied into my apps _build dir. 

I also ran across this [issue](https://github.com/bitwalker/exrm/issues/44) while not directly related I did note that the there was an issue with erlang 17.1 (seems like I remember that actually) according to that dealing with sym links. I did update my raspberry pi and should be the lastest 18 release (don't remember the minor versions), but I do remember it being 17 something before I updated it last night.

When I get a chance let me make sure that isn't the root cause, mix might be using the wrong Erlang version.

fhunleth, 2015-11-24:
I'm pretty sure that I found the problem. Elixir-ALE's Makefile creates the `priv` directory the first time that it's run. That seems to confuse `mix` since it seems to only create the symlink if the `priv` directory exists beforehand and not afterwards. If you compile a second time, then everything works. I updated the `master` branch with a dummy file so that the `priv` directory is always there. That fixes this issue for me. Let me know if this fixes it for you, and thank you so much for bringing this issue to my attention.

matthewphilyaw, 2015-11-24:
That change worked. Thanks for looking into it! That's so strange, in fact was looking into it my self and noted that if I went into your project in the deps folder and called `mix` it would generate a _build directory with a sym link to priv. So it worked at that level. So mix will sym link properly, but not for the dependencies.

I wonder if this is a bug in mix? It seems weird that it works correctly if you build your project by it self using mix, but not as a deps.

matthewphilyaw, 2015-11-24:
Sorry, should correct my self misread your comment and indeed even locally if I build your project standlone if the priv isn't there the local _build doesn't have it either. 

Again thanks for looking into it, just odd behavior but I guess it makes sense.

fhunleth, 2015-11-24:
I wasn't sure whether it was a bug in `mix` or not, but at least now `elixir_ale` works again. I'll be pushing an update to hex shortly. Thanks again for testing.

closed 2015-11-24 by fhunleth

josevalim, 2015-11-24:
Btw, @matthewphilyaw, what is your Elixir version? We had a similar issue in the past which is supposedly fixed in more recent Elixir versions.

matthewphilyaw, 2015-11-24:
@josevalim 

Erlang/OTP 18 [erts 7.0]
Elixir 1.1.1

Should I log this as an issue there? I went to log the issue but found https://github.com/elixir-lang/elixir/issues/3728, which was closed but sounded very similar didn't want to create dup issues.

josevalim, 2015-11-24:
Good call @matthewphilyaw. The fix to that particular issue was this one:

https://github.com/elixircnx/comeonin/pull/41

I would say elixir_ale should do the same, otherwise whatever is built on priv won't be copied on platforms like Windows.

paulanthonywilson, 2016-01-01:
I have just run into what seems to be the same issue using 0.4.0. From a clean build, the `ale` binary is built in `deps/elixir_ale/priv/` but it did not make it into `_build/prod/lib/elixir_ale/priv/` and subsequently as missed from the release.

Subsequent compilation did not fix the issue  as the dependency was already built, as far as Mix was concerned. 

However `rm -rf _build/prod/lib/elixir_ale/` and compiling again did solve the problem. Following @fhunleth's suggestion in Slack, using Elixir Ale master also solves the issue.

Environment:

* Elixir 1.1.1
* OS X 10.10.5
* Using `brew nerves` for Raspberry PI 2

matthewphilyaw, 2016-01-01:
the hex package 0.4.0 doesn't contain the change in ee67c5f if I'm not mistaken, but master does.

fhunleth, 2016-01-01:
I just published v0.4.1 here and on hex. The fix should be generally available now.

paulanthonywilson, 2016-01-01:
Nice! That was quick :-)

josevalim, 2016-01-01:
:heart: :green_heart: :blue_heart: :yellow_heart: :purple_heart:

fhunleth, 2016-01-03:
@paulanthonywilson You caught me at a particularly convenient time to make the release! Thanks for alerting me that the previous workaround in v0.4.0 for this issue wasn't working for you.