fhunleth/elixir_ale #9: Trying to compile on Mac

gordonbisnor, 2016-02-16:
I'm trying to add elixir ale as a dependency in an Elixir/Phoenix project on a Mac but have been unsuccessful.

I first got errors about ei.h and set this in the Elixir/Ale Makefile:

```
ERL_EI_INCLUDE_DIR = /usr/local/lib/erlang/usr/include
```
I then got an error about libei.a, and set this in the Makefile:

```
ERL_EI_LIBDIR = /usr/local/lib/erlang/usr/lib
```

Now I'm stumped on this error:

```
cc -c -I/usr/local/lib/erlang/usr/include -O2 -Wall -Wextra -Wno-unused-parameter -o src/i2c_port.o src/i2c_port.c
src/i2c_port.c:33:10: fatal error: 'linux/i2c.h' file not found
#include <linux/i2c.h>
         ^
1 error generated.
make: *** [src/i2c_port.o] Error 1
could not compile dependency :elixir_ale, "mix compile" failed. You can recompile this dependency with "mix deps.compile elixir_ale", update it with "mix deps.update elixir_ale" or clean it with "mix deps.clean elixir_ale"
```

Wondering if I should I be using a Docker container if I want to develop on Mac, or etc....

fhunleth, 2016-02-16:
Elixir ALE only supports the Linux GPIO, I2C, and SPI interfaces, so it will only work on Linux systems. Most people use elixir_ale on Raspberry Pis or Beaglebones or other embedded boards rather than laptops or desktop PCs. 

What are you trying to do?

gordonbisnor, 2016-02-16:
I'm just starting to learn about Nerves and Raspberry Pi, and have been developing Elixir applications with Nerves on Mac and then using bake and fwup to get them onto the SD card for Raspberry Pi.

fhunleth, 2016-02-16:
Got it. With Nerves, you'll be cross-compiling Elixir ALE for the Raspberry Pi. Bake helps with that. My recommendation, though, is to install Elixir on Raspbian on your Raspberry Pi and develop there. Once you get comfortable with Elixir and Elixir ALE on the RPi, then switch to Nerves. Nerves is good at the packaging apps to run standalone, but not good at being a friendly development environment yet.

closed 2016-02-16 by fhunleth

gordonbisnor, 2016-02-16:
Thanks for the advice!