fhunleth/elixir_ale #11: Missing Erlang Header Files

Zanadar, 2016-04-14:
I'm getting the following error when I compile on a RaspberryPi 2B:

```
warning: the VM is running with native name encoding of latin1 which may cause Elixir to malfunction as it expects utf8. Please ensure your locale is set to UTF-8 (which can be verified by running "locale" in your shell)
Makefile:17: *** Could not find include directory for ei.h. Check that Erlang header files are available.  Stop.
** (MatchError) no match of right hand side value: 2
    mix.exs:4: Mix.Tasks.Compile.ElixirAle.run/1
    (elixir) lib/enum.ex:1043: anonymous fn/3 in Enum.map/2
    (elixir) lib/enum.ex:1387: Enum."-reduce/3-lists^foldl/2-0-"/3
    (elixir) lib/enum.ex:1043: Enum.map/2
    (mix) lib/mix/tasks/compile.all.ex:19: anonymous fn/1 in Mix.Tasks.Compile.All.run/1
    (mix) lib/mix/tasks/compile.all.ex:37: Mix.Tasks.Compile.All.with_logger_app/1
    (mix) lib/mix/tasks/compile.ex:64: Mix.Tasks.Compile.run/1
    (mix) lib/mix/cli.ex:55: Mix.CLI.run_task/2
```
Probably related to #https://github.com/nerves-project/nerves-toolchain/issues/3 . I'll investigate further...

fhunleth, 2016-04-14:
If I remember right, on Raspberry Pi this is due to not installing the erlang package with the header files. Maybe `sudo apt-get install erlang-dev`? Could you check if `ei.h` exists on your system?

Zanadar, 2016-04-14:
Indeed I looked and `ei.h` isn't there. I'll try some other installation options and report findings.

Zanadar, 2016-04-14:
I left a small PR with a note in the readme on my solution to this issue (https://github.com/fhunleth/elixir_ale/pull/12).

fhunleth, 2016-04-14:
I merged the PR. Thanks for sending it.

closed 2016-04-14 by fhunleth