fhunleth/elixir_ale #16: Reading from GPIO on a Raspberry Pi 3 Model B

hzeus, 2016-06-17:
I am trying read key presses from a GPIO from a Raspberry Pi 3. I am currently using Raspbian for it. My problem is that I cannot detect the key presses, as `Gpio.read` always returns 0.

I am using the master branch (commit 291005bcf) and try inside `iex -S mix`

```elixir
{:ok, pid}=Gpio.start_link(18, :input)
# => {:ok, #PID<0.1044.0>}
Gpio.read(pid)
# => 0

# When Button is pressed
Gpio.read(pid)
# => 0
```

When I try the same setup with the python script (run via the preinstalled python, without sudo privileges)

```python
import RPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.IN, pull_up_down=GPIO.PUD_UP)

while True:
    if GPIO.input(18) == False:
        print('Button pressed')
    time.sleep(0.1)
```

I can see the "Button pressed" output whenever I press the button.

Finally, when doing the exact same thing as above after I have run the python script it works:

```elixir
{:ok, pid}=Gpio.start_link(18, :input)
# => {:ok, #PID<0.120.0>}
Gpio.read(pid)
# => 1

# When Button is pressed
Gpio.read(pid)
# => 0

# When Button is released
Gpio.read(pid)
# => 0
```

I assume the python script is setting something up that I am missing in elixir? 

As I wrote above, I am currently using Raspbian, but my initial goal was to use the nerves project. When I run the same code from an iex inside nerves, I experience the same behavior as above (`Gpio.read` always returns 0). Do you have any suggestions? (As well as for Raspbian as for nerves)?

fhunleth, 2016-06-17:
I assume that you have the button connected without a pullup resistor. The Python version enables an internal pullup (inside of the RPI's CPU) to ensure that the GPIO reads high when the button isn't pressed. Without the pullup, the GPIO could read anything. To my knowledge, configuring the internal pullups/pulldowns on Raspbian is still platform-specific. Since I haven't had much time to maintain this project, I haven't worked on platform-specific features. 

If you have a 10K ohm resistor (or anything around that value), try connecting one side to the GPIO and one side to +3.3V. That will fix your problem by making sure that the GPIO is either forced high when the button isn't pressed or low when it is.

If you want to look into how the Python library sets the pullups and pulldowns and see if that can be ported to ALE, that would be useful too. A few weeks ago someone expressed interested in adding support for this, but I can't find their handle at the moment. Maybe they will see this and respond.

hzeus, 2016-06-18:
Thank you for your very fast response! I will buy a resistor some time during the following week; and I will then answer whether I succeeded.

hzeus, 2016-06-23:
I tried adding a pullup resistor and it worked. In case any other newbie stumbles over this issue: I found http://www.bit-101.com/blog/?p=3813 to be a good explanation of what these are and how to apply them.

closed 2016-06-23 by hzeus

TheBerg, 2016-08-28:
@fhunleth I forgot my resistors at work, so I decided to play around with seeing if I can make this happen. My C fu, and frankly my Elixir fu isn't that great. But I did see how they got this working in the ruby library. They bundle & import the [libbcm2835.so library](https://github.com/jwhitehorn/pi_piper/blob/v2.0.0/lib/pi_piper/bcm2835.rb#L8) then they pull in a function called [`pin_set_pud` from that library](https://github.com/jwhitehorn/pi_piper/blob/v2.0.0/lib/pi_piper/bcm2835.rb#L31).  

Is this something we could try to do for elixir_ale?

fhunleth, 2016-08-28:
@TheBerg The problem with including this in elixir_ale is that it will only work for the Raspberry Pi. I'm keeping elixir_ale generic, since we learned a couple years ago that things get really hard to maintain when platform-specific pieces get included. 

Having said that, it would be great if someone could build a Raspberry Pi I/O library for Elixir that supports pullups, PWM, and other RPi things. Maybe something similar in functionality to gpio_zero (a Raspberry Pi-only library). If anyone else is reading this and is interested, please know that many people would use your library including me.