fhunleth/elixir_ale #20: Allow larger I2C buffers than 32 bytes

bendiken, 2016-08-16:
Just ran into `ale: write: need a binary between 1 and 32 bytes` when calling `I2c.write/2` with a 96-byte payload.

I see that this error is originating from https://github.com/fhunleth/elixir_ale/blob/f8d3fe72407eb068a67cbf5c79f9d6f16726c4e7/src/i2c_port.c#L152

Is there any particular reason that I2C writes are limited to `I2C_SMBUS_BLOCK_MAX`, which is 32 bytes? SMBus as a subset of I2C is limited to 32 bytes, but I2C per se has no such limitation. It is a problematic limit for dealing with larger buffers such as with an OLED display.

fhunleth, 2016-08-16:
We should definitely allow longer transactions. Unless the temporary buffer that holds the data is moved off the stack, I wouldn't want the max size to be too large, but 256 bytes or a little more would be completely reasonable. 

Could you try it out with the higher limit to verify that your OLED works? 

Thanks!

bendiken, 2016-08-17:
Submitted pull request to increase the size to 512 bytes, a value likely to be suitable for most uses (incidentally, the same size as `PIPE_BUF` on most POSIX platforms). Tested with my OLED driver, works fine.

bendiken, 2016-08-18:
Closing this as #21 has been merged.

closed 2016-08-18 by bendiken