fhunleth/elixir_ale #23: Wire Protocol: 1-wire

coop, 2016-09-03:
Hi,

This is probably the wrong place for this kind of request but I was talking with @GregMefford in #nerves this evening and he thought it would be a reasonable starting point.

>If you need to implement a wire level protocol to talk to a device, look for a Linux kernel driver. It may just be a matter of loading the right kernel module.

I'm wanting to take temperature readings from k type thermocouple via a MAX31850 on my raspberry pi 3 - given what you've said above I think I need to reach out for a linux driver. I modified my `config.txt` file and added support for dtoverlay:

```
+ # Comment this in or modify to enable OneWire
+ # NOTE: check that the overlay that you specify is in the boot partition or
+ #       this won't work.
+ dtoverlay=w1-gpio-pullup,gpiopin=4
```

I was then able to see my device and cat `w1_slave` - unfortunately the result is:

```
00 00 00 00 00 00 00 00 00 : crc=00 NO
00 00 00 00 00 00 00 00 00 : t=0
```

Just wondering if this is a problem `elixir_ale` would be looking to support or should I go elsewhere?

NOTE: Similar solution with rp and python - https://github.com/steve71/RasPiBrew/blob/master/RasPiBrew/Temp1Wire.py

Thanks

fhunleth, 2016-09-03:
I've actually never used a OneWire device. I know that a couple other people have gotten OneWire working on Nerves, but I'm not sure what they did differently. Given that you're getting all zeros, I'd assume that the pullup isn't strong enough, since you never get a 1, but that's just guessing.

The second question about including OneWire support in elixir_ale or not, my preference would be to start something outside of elixir_ale to see how it goes. If the API is really simple like I2C and SPI, then I'd think that it makes sense to include.

ulve, 2016-09-04:
I've used a ds18b20 one wire temperature sensor from elixir on a rpi2. Never tried nerves or ale though. I could have a stab at the ds18b20 using nerves if you like @coop ? not the same thing but both using one wire.

coop, 2016-09-04:
@ulve I'd appreciate any help I can get. Ale might not make sense (I'd probably even encourage us to start the project elsewhere) but it is a good enough spot to have the discussion considering it is closely related. 

My problem is that I don't know what "I'd assume that the pull-up isn't strong enough" means to even start debugging the issue. 

Thanks

coop, 2016-09-09:
Hey @ulve - appreciate you staying in touch.

I found [your project on GitHub](https://github.com/ulve/ds18b20_elixir/blob/df9fd42cfefe9a6d0f32d12174d89f26fe70e843/lib/reader.ex) - it is exactly what I expected. List the devices and report the temps. Should be pretty easy.

Unfortunately I'm still stuck, I saw your comment about using a physical resistor:

>I used a 4.7kohm resistor when measuring temperature with my 18ds20

I've done this also. I actually have this project working using a Particle Photon, I'm just interested how an elixir version would stack up. I've currently wired it according to this [adafruit tutorial](https://learn.adafruit.com/adafruit-1-wire-thermocouple-amplifier-max31850k/wiring-and-test) - Photon works fine, Nerves + pi3 nada.

Over the weekend I'm going to see if it works with raspberian - I'm hoping that will narrow down the places to look.

Thanks again.

coop, 2016-09-09:
FYI I changed out the provided 4.7k resistor for a 10k and the device wouldn't register.

coop, 2016-09-09:
Ok, I have downloaded and installed raspberian using the instructions on their website. Initially the devices didn't show up so I made 2 modifications:

```
$ sudo raspi-config
# enabled 1 wire
```

This wasn't enough by itself, I also had to modify `/etc/modules`:

```
$ cat /etc/modules
# existing comments ...
w1-therm
```

After a `reboot` my device started showing up but reading from it only produced zeros:

```
00 00 00 00 00 00 00 00 00 : crc=00 NO
00 00 00 00 00 00 00 00 00 t=0
```

I think it is positive that I'm getting the same result in nerves and raspberian - leads me to believe @fhunleth has been right all along:

>Given that you're getting all zeros, I'd assume that the pullup isn't strong enough, since you never get a 1, but that's just guessing.

I will keep investigating.

GregMefford, 2016-09-10:
This is where it would be really handy to have an oscilloscope to watch the voltage on the wire and see if it's even responding. I was thinking maybe since there's only one wire involved, the Pi wasn't switching from output mode to input mode at the right time. If that were the case, though, I wouldn't think it could detect the device and read its address and stuff.

I don't believe I have any 1Wire hardware I can test with myself, unfortunately.

jmedding, 2016-09-16:
I landed here because I also wanted to add DHT11 (a temperature and humidity sensor that communicates via a one-wire protocol). I have no idea if it's the same protocol though.  For the API, I had in mind two variants.

1. Poll: You ping the sensor and get the current value. The underlying C code would print the results as a string to stdout and exit.

2. Update: The sensor is read at an interval (configurable) which updates the state, which can be read from other processes at any time, giving the last value.

3 Emit: Like 2, but each time there is a new value, it is broadcast over a pub-sub. 

I thought to use genservers for (1,2) but I'm not sure what is the best mechanism for (3). GenEvent is not great, because the events stay inside their own process. Genstage is interesting, but as it's a pull model I don't think its the right thing either (would work for 1 and 2 though).

I'm not a C programmer, but it looks like Java, which I knew 10 years ago, so I will try and get something going as an example.

tverlaan, 2017-05-19:
Hi, I just saw this thread and I've been using 1Wire for quite some time now. You need the 'hack' as mentioned in the first comment and from there on you can use this library: https://github.com/mokele/onewire_therm

Working example: https://github.com/tverlaan/brew

I don't see an immediate reason to add this to elixir_ale, but I guess it would be a nice addition for people just starting. Maybe a link in the readme will suffice?

commit c1b4b6a2 (2018-11-02): Add FAQ entry for 1-wire, closes #23
closed 2018-11-02 by fhunleth