fhunleth/elixir_ale #29: Warning when compiling on Raspberry Pi

CraigCottingham, 2016-11-20:
Erlang/OTP 19 [erts-8.1] [source] [async-threads:10] [hipe] [kernel-poll:false]
Elixir 1.3.4
Raspberry Pi B+ v1.2
Raspbian Jessie/Pixel

When compiling `elixir_ale` (via `iex -S mix` after adding it as a dependency), I get these compiler warnings.

```
src/spi_port.c: In function ‘spi_transfer’:
src/spi_port.c:115:19: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
     tfer.tx_buf = (__u64) tx;
                   ^
src/spi_port.c:116:19: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
     tfer.rx_buf = (__u64) rx;
                   ^
```

I'm leaving this as much as a note for myself as anything else. My C skills are rusty, but I'll try to remember to circle back and address this, if someone else doesn't first. 😄

fhunleth, 2016-11-21:
That's a really unfortunate error message. The Linux header file for making SPI transfers requires you to pass a pointer in a 64-bit unsigned integer variable (the `__u64`). However, pointers on the Raspberry Pi are 32-bits so you get the warning. Since I can't change the Linux header file, there's really no way for this to be fixed other than by turning off that one warning for those lines of code. 

If you want to give it a try, could you see if adding the following above line 115:
```
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wpointer-to-int-cast"
```
and the following below line 116:
```
#pragma GCC diagnostic pop
```
If it works, could you add a short comment and send a PR? I have no doubt that others will take a second look at the warning when they see it.

CraigCottingham, 2016-11-21:
Ugh. It annoys me to have to disable warnings, but I get why it's necessary in this case. I'll give this a try later tonight.

fhunleth, 2016-11-22:
Yes, I completely agree. Thanks for giving it a try, though.

fhunleth, 2016-11-23:
Fixed per #30.

closed 2016-11-23 by fhunleth