fhunleth/elixir_ale #31: different ei.h paths

BenBals, 2016-12-01:
I installed my Erlang via homebrew. The Erlang stuff, including ei.h is in `/usr/local/Cellar/erlang/19.0.2/lib/erlang/`. How can I edit the Makefile to include it as a search path?

Thanks in advance.

fhunleth, 2016-12-01:
It's possible, but you'll get an error shortly after that since your Mac won't have any of the I2C or SPI libraries on it and the GPIO code won't work at runtime. If you have a Raspberry Pi or BeagleBone, I'd recommend building it there or trying out Nerves. If you're already doing that and just want to test on your laptop, most people stub out Elixir ALE. See the FAQ in the README for a good link.

BenBals, 2016-12-01:
I am using Nerves. What I am trying to do is compiling my Nerves project on
my Mac. I have the crosscompiler and set the flag. Now the Makefile can't
find ei.h because Homebrew installed into a folder different from the paths
searched by the Makefile. I know where it is. How can I tell it to search
the right path?

fhunleth, 2016-12-01:
Then something else is wrong since ei.h needs to come from the version of Erlang that's going to your target. That comes from the Nerves system image. Another way to think about this is if the C code gets compiled using libraries compiled for OSX (coming from Homebrew), it won't work on your RPi, BBB, or whatever you're using. 

I'm going to be offline for the next 9 hours or so. You might have some luck asking on the nerves channel on the Elixir-lang slack. What you're doing should work out of the box, so I'm a little surprised. Perhaps @mobileoverlord has an idea.

mobileoverlord, 2016-12-01:
@BenBals do you have an example project which mimics the issue you are having. The Nerves Env should take care of setting the ERL_EI_INCLUDE_DIR and ERL_EI_LIBDIR env variables.

The Makefile is currently laid out to only set them by looking in the search paths if they are not already set `?=`

You can see some additional information regarding your Nerves Env by putting `NERVES_DEBUG=1 mix compile`

BenBals, 2016-12-01:
Doing a clean git clone fix the problem. Don't know what was wrong. Thanks for your kind help.

BenBals, 2016-12-01:
Now this is super weird. My app is an umbrella project. I can do `mix test`, `mix compile`, `mix firmware` and so on in `apps/firmware` without any problems, but I can't run `mix test` from the project root. The project is open source at https://gitlab.com/vitabox/vitaprog.

mobileoverlord, 2016-12-01:
When commands are executed at the root of an umbrella they are executed in the scope of the umbrella mix environment. This means that the aliases that are defined in your `apps/firmware` project are never invoked, which means Nerves Env is never invoked and it's attempting to run under OS X.

To run test from the top level, use `mix cmd mix test` This will enumerate and execute each app in the umbrella in its own environment.

BenBals, 2016-12-01:
Thank you. That solves the problem.

closed 2016-12-02 by fhunleth