fhunleth/elixir_ale #32: Increase i2c max buffer size to system default.

jamesotron, 2017-01-01:
Following up on #20, I'm also driving an OLED display talking to an SSD1306 [code here](https://github.com/jamesotron/ssd1306.ex) and running into performance issues because when sending data to the display (128 x 64 pixels, so 1KB buffer) I have to send 1 byte with each transaction to tell the chip to write it into the display buffer, meaning I can send a maximum of 511 bytes at a time and have to make three seperate calls to the display.

Anyway, all this led me down the rabbit hold of researching Linux's i2c implementation, and it looks like the absolute maximum buffer size is 8KB as defined here:

https://github.com/raspberrypi/linux/blob/rpi-4.0.y/drivers/i2c/i2c-dev.c#L168

Other than changing a single constant in the source, is there an issue with increasing the buffer size to 8k?

fhunleth-lkc, 2017-01-02:
Hey - sorry for the delay, I'm majorly catching up from being offline the past week. Updating the max buffer size to 8K seems reasonable to me. Thanks for tracking down the max Linux size. I didn't know that it had grown so large.

fhunleth, 2017-04-03:
Closing since @adkron sent a PR that fixes it.

closed 2017-04-03 by fhunleth