fhunleth/elixir_ale #35: Cross Compilation on Mac fails

lucca65, 2017-03-15:
Hi, after reading #9 I've tried to cross compile `elixir_ale` for my nerves project, I get the following error:

```shell
λ ESPX grow → λ git master* → CROSSCOMPILE=1 mix deps.compile elixir_ale
==> elixir_ale
Makefile:24: *** Could not find include directory for ei.h. Check that Erlang header files are available.  Stop.
```

Seems like `elixir_ale` is trying to find some headers, checked the `Makefile` and found this:

```shell
find /usr/local/lib/erlang /usr/lib/erlang ~/.asdf/installs/erlang/19.0/lib/erlang/usr/include/ -name ei.h -printf '%h\n' 2> /dev/null | head -1
```

When trying to run this, without `|head` i get the following:

```shell
λ ESPX elixir_ale → λ git master* → find /usr/local/lib/erlang /usr/lib/erlang ~/.asdf/installs/erlang/19.0/lib/erlang/usr/include/ -name ei.h -printf '%h\n'             
find: -printf: unknown primary or operator
```

Maybe `%h` doesn't work the same way in mac? I'm in the right direction here?

fhunleth, 2017-03-15:
I'm sure that you're headed in the right direction and that the Mac has the BSD version of `find` rather than the GNU one. Could you try `brew install findutils` and replace `find` with `gfind`?

lucca65, 2017-03-15:
Thanks, the gnu version worked fine, but got stuck in the same problem #9 pointed out. 

There is no way to make sure `elixir_ale` compiles it inside the toolchain images for nerves?

fhunleth, 2017-03-15:
If you're using Nerves, then this logic isn't run if you have your target set. Nerves sets the environment variables for you. I thought that you were trying to crosscompile outside of a Nerves environment. You can manually source the crosscompile environment that Nerves uses, but I think your best bet is to go with the more straightforward route of letting Nerves do it for you. You may want to try out one of the https://github.com/nerves-project/nerves-examples and check out the nerves channel on the elixir-lang slack for interactive support.

closed 2017-03-15 by fhunleth

lucca65, 2017-03-15:
I see, my project was already working, but with nerves update, it stopped, figured out it they've changed some interaction with `elixir_ale`. I will try to move my code to a fork of those examples! Thanks for your help 😄