fhunleth/elixir_ale #43: i2c_port.c, line 134, Size of resp[] should be increased

mdsebald, 2017-04-05:
I2C_BUFFER_MAX has been increased to 8K.  Don't know if it's true, but the comment on line 134, indicates the size of resp[] should be increased, also:

`char resp resp[1024]; // Must be larger than I2C_BUFFER_MAX`

fhunleth, 2017-04-05:
Yes, you're right. I totally forgot about that when I merged it in. Thanks for catching this! I don't remember what the message overhead is, but I think that it's on the order of 10s of bytes max so an 8300 byte buffer should be sufficient. This requires more scrutiny, though, and seems like the allocation should be cleaned up to actually be calculated from I2C_BUFFER_MAX. 

Not sure if you want to look at it, but if you don't get to it, I'll check it out this weekend or early next week.

mdsebald, 2017-04-05:
Unfortunately, I don't use elixir_ale, so I wouldn't be able to test any changes.  I noticed this when I was comparing it to erlang_ale.

fhunleth, 2017-04-09:
I have a fix for this queued up in a branch: https://github.com/fhunleth/elixir_ale/tree/fix-buffer-size.

I haven't tested on real hardware, but as soon as I do that, I'll pull it into `master`.

fhunleth, 2017-04-12:
Should be fixed in 44c26ee40243efd2b39da2140f0a705ed288eb79. I don't have anything that approaches the max I2C size to test this.

closed 2017-04-12 by fhunleth