fhunleth/elixir_ale #53: Concurrent reads crash the Port

sandromehic, 2018-02-22:
Hello, first of all thanks for this library that allows me to use Elixir for some IoT applications that I am working on. The issue that I am reporting has to do with concurrent read accesses to the SPI device. When more than one message arrives to the `ElixirALE.SPI` GenServer at the same moment, the GenServer will crash because the `Port` returns an `:error` code. This could even be a `Port` or C library issue, I don't have enough knowledge to debug that part.

As I understand the `GenServer` behaviour the messages should be handled sequentially, so the previous operation should have finished by the time the second call is executed internally, but evidently this is not the case. I have tried also modifying the `:delay_us` parameter since it seemed to the one to control the delay between SPI transmission, but with no changes in the outcome. 

Is there some design pattern that I am missing here that would make sure that the GenServer receives only one call at a time? Is there checks that could be done in the library that would fix this (in case, I can work on a PR)?

Here is an example of how to reproduce this problem. In a new mix project with supervisor, I start the `ElixirALE.SPI` GenServer with name `ElixirALE.SPI.Server`.

```elixir
defmodule Example.Application do
  @moduledoc false

  use Application

  def start(_type, _args) do
    children = [
      %{
        id: ElixirALE.SPI,
        start: {ElixirALE.SPI, :start_link, ["spidev0.0", [], [name: ElixirALE.SPI.Server]]}
      }
    ]

    opts = [strategy: :one_for_one, name: Example.Supervisor]
    Supervisor.start_link(children, opts)
  end
end
```
Define a module that does sequential and concurrent access to the GenServer through its API:

```elixir
defmodule Example do
  def transfer, do: ElixirALE.SPI.transfer(ElixirALE.SPI.Server, <<0x00, 0x00, 0x00>>)
  def task(_), do: Task.async(fn -> transfer() end)

  def sequential_ok do
    0..2
    |> Enum.each(fn _ -> transfer() end)
  end

  def concurrent_ko do
    0..2
    |> Enum.map(&task/1)
    |> Enum.map(&Task.await/1)
  end
end
```
Now from `iex -S mix` we can see that sequentially calling `ElixirALE.SPI` transfer won't crash:
```elixir
iex> Example.sequential_ok
:ok
```
If the `transfer` function, that does a `GenServer.call`, is called concurrently, the `GenServer` crashes since the `Port` does not return a `{_, {:data, response}}` response, but something else.
```elixir
iex> Example.concurrent_ko
** (EXIT from #PID<0.146.0>) shell process exited with reason: exited in: GenServer.call(ElixirALE.SPI.Server, {:transfer, <<0, 0, 0>>}, 5000)
    ** (EXIT) an exception was raised:
        ** (MatchError) no match of right hand side value: :error
            (elixir_ale) lib/elixir_ale/spi.ex:109: ElixirALE.SPI.handle_call/3
            (stdlib) gen_server.erl:615: :gen_server.try_handle_call/4
            (stdlib) gen_server.erl:647: :gen_server.handle_msg/5
            (stdlib) proc_lib.erl:247: :proc_lib.init_p_do_apply/3

Interactive Elixir (1.6.1) - press Ctrl+C to exit (type h() ENTER for help)
iex(1)>
20:32:16.713 [error] GenServer ElixirALE.SPI.Server terminating
** (MatchError) no match of right hand side value: :error
    (elixir_ale) lib/elixir_ale/spi.ex:109: ElixirALE.SPI.handle_call/3
    (stdlib) gen_server.erl:615: :gen_server.try_handle_call/4
    (stdlib) gen_server.erl:647: :gen_server.handle_msg/5
    (stdlib) proc_lib.erl:247: :proc_lib.init_p_do_apply/3
Last message: {:transfer, <<0, 0, 0>>}
State: %ElixirALE.SPI.State{devname: "spidev0.0", port: #Port<0.4374>}

20:32:16.720 [error] Task #PID<0.148.0> started from #PID<0.146.0> terminating
** (stop) exited in: GenServer.call(ElixirALE.SPI.Server, {:transfer, <<0, 0, 0>>}, 5000)
    ** (EXIT) an exception was raised:
        ** (MatchError) no match of right hand side value: :error
            (elixir_ale) lib/elixir_ale/spi.ex:109: ElixirALE.SPI.handle_call/3
            (stdlib) gen_server.erl:615: :gen_server.try_handle_call/4
            (stdlib) gen_server.erl:647: :gen_server.handle_msg/5
            (stdlib) proc_lib.erl:247: :proc_lib.init_p_do_apply/3
    (elixir) lib/gen_server.ex:830: GenServer.call/3
    (elixir) lib/task/supervised.ex:88: Task.Supervised.do_apply/2
    (elixir) lib/task/supervised.ex:38: Task.Supervised.reply/5
    (stdlib) proc_lib.erl:247: :proc_lib.init_p_do_apply/3
Function: #Function<2.89360649/0 in Example.task/1>
    Args: []

20:32:16.721 [error] Task #PID<0.150.0> started from #PID<0.146.0> terminating
** (stop) exited in: GenServer.call(ElixirALE.SPI.Server, {:transfer, <<0, 0, 0>>}, 5000)
    ** (EXIT) an exception was raised:
        ** (MatchError) no match of right hand side value: :error
            (elixir_ale) lib/elixir_ale/spi.ex:109: ElixirALE.SPI.handle_call/3
            (stdlib) gen_server.erl:615: :gen_server.try_handle_call/4
            (stdlib) gen_server.erl:647: :gen_server.handle_msg/5
            (stdlib) proc_lib.erl:247: :proc_lib.init_p_do_apply/3
    (elixir) lib/gen_server.ex:830: GenServer.call/3
    (elixir) lib/task/supervised.ex:88: Task.Supervised.do_apply/2
    (elixir) lib/task/supervised.ex:38: Task.Supervised.reply/5
    (stdlib) proc_lib.erl:247: :proc_lib.init_p_do_apply/3
Function: #Function<2.89360649/0 in Example.task/1>
    Args: []
```
thanks

fhunleth, 2018-02-23:
I'm surprised that it crashed. While I don't think that most people have multiple processes calling into ElixirALE.SPI.transfer, that shouldn't have been a problem. Fwiw, the normal route is to implement a module that abstracts the details of how to encode messages to the device and possibly also to keep some device state and have all of your application code call into that. However, I'd like to figure out what's going on here. What board are you running on?

sandromehic, 2018-02-23:
Yeah, it suprised me too since I was expecting the GenServer to process the calls in a manner that the port is available between calls. I am working on a RPi3 board with a MCP3008 ADC in this moment, and have a module that I use for accessing the `ElixirALE.SPI` GenServer like this:
```elixir
defmodule Example.SPIRead.MCP3008 do

  @spi_server ElixirALE.SPI.Server
  @start_byte 0x01
  @channels %{
    0 => 0x80,
    1 => 0x90,
    2 => 0xA0,
    3 => 0xB0,
    4 => 0xC0,
    5 => 0xD0,
    6 => 0xE0,
    7 => 0xF0
  }
  @end_byte 0x00

  @doc """
  Reads the channel `n` from the MCP3008 ADC and returns the result.

  Expects a `GenServer` of `ElixirALE.SPI` to have been started with
  @spi_server name.
  """
  def channel(n) when n < 0, do: :rand.uniform()

  def channel(n) do
    <<_::size(14), value::size(10)>> =
      ElixirALE.SPI.transfer(@spi_server, <<@start_byte, @channels[n], @end_byte>>)

    value
  end
end
```

I am thinking about introducing a lock in this module so that new calls are sent to `ElixirALE.SPI.Server` only when previous was processed. It would be interesting to figure out how come this thing is crashing, since this lock should not be necessary since there is already a GenServer that should think about handling the calls.

If you need more debug information, I can try it with my setup and provide more feedback.

sandromehic, 2018-02-23:
Currently, sovled my issue by with [Mutex](https://hexdocs.pm/mutex/Mutex.html) and requiring a `lock` when `ElixirALE.SPI.transfer` is called.

This solves the problem with concurrent access, but I'm still not sure how come it's not working with the already existing `GenServer` that should make sure to execute the calls in order and sequentially.

For anybody with a similar issue, this is how I changed my `Example.SPIRead.MCP3008` module to make it work (note the lock acquisition and release):

```elixir
defmodule Emagex.SPIRead.MCP3008 do

  @spi_server ElixirALE.SPI.Server
  @mutex ElixirALE.SPI.Mutex

  @start_byte 0x01
  @channels %{
    0 => 0x80,
    1 => 0x90,
    2 => 0xA0,
    3 => 0xB0,
    4 => 0xC0,
    5 => 0xD0,
    6 => 0xE0,
    7 => 0xF0
  }
  @end_byte 0x00

  @doc """
  Reads the channel `n` from the MCP3008 ADC and returns the result.

  Expects a `GenServer` of `ElixirALE.SPI` to have been started with
  @spi_server name.
  """
  def channel(n) when n < 0, do: :rand.uniform()

  def channel(n) do
    lock = Mutex.await(@mutex, :spi)
    <<_::size(14), value::size(10)>> =
      ElixirALE.SPI.transfer(@spi_server, <<@start_byte, @channels[n], @end_byte>>)

    Mutex.release(@mutex, lock)
    value
  end
end
```

@fhunleth  I don't know if you prefer to close this issue, or investigate further. Let me know if you need more info!

cdegroot, 2018-02-23:
(flyby comment, feel free to ignore) my guess is that the receive in https://github.com/fhunleth/elixir_ale/blob/master/lib/elixir_ale/spi.ex#L122 receives the second genserver call that's in the mailbox and then errors out. I think a more fine-grained error match should happen there (stuff that would actually come back from the port) so that the second genserver call is left in the mailbox until the whole current operation is done.

sandromehic, 2018-02-23:
@cdegroot thanks for the heads up, it was actually an issue with using a `receive` block in a `GenServer` that was intercepting the concurrent calls to `GenServer`.

I have made a pull request #54 with the fix that uses `handle_info` callbacks instead of `receive` (as suggested by [`GenServer` docs](https://hexdocs.pm/elixir/GenServer.html#module-receiving-regular-messages)). To make it work the `GenServer` needs to keep in state if it is waiting to reply to someone, and in that case it will keep trying the concurrent calls until the `Port` has responded.

fhunleth, 2018-03-02:
Fixed in v1.0.3. Thanks again for the report and verifying the fix!

closed 2018-03-02 by fhunleth