fhunleth/elixir_ale #55: Pull port calls to behaviors and their own modules.

adkron, 2018-02-25:
The calls to the ports are prohibitive to testing. I suggest creating a behaviour and moving the calls to their modules. We can then use the behaviour with Mox for testing. Right now we can do some rudimentary testing by sending the same messages to the GenServers that the ports would send. Unfortunately, that would not be able to find bugs like #54 because we need to send data to the port. Using mocks gives fine-grained control over the responses, and lets us test on systems without actually needing a device that supports the protocols.

fhunleth, 2018-03-02:
I need to think about this more, but I have a few comments:

1. I desperately want easier testing. Elixir 2.0 is in the works, so I'm thinking that if there are any API changes that we push it in the 2.0 branch to avoid noise for 1.x users.
2. There are two ways to look at testing: testing the guts of the library and helping people test their code that uses Elixir ALE.
3. For testing the guts of this library, I'd like to exercise as much C code as possible. My thought on doing this is to add a "loopback" mode for GPIO, I2C, and SPI. This mode could be entered via a special bus or GPIO. The C side would need a little extra work to handle it, but I think this would catch the main issues that I've seen with maintaining ElixirALE over the past few years. I don't see an easy way of using real hardware with automated tests. 
4. For helping library users test their code, I think that Mox sounds like the right way. I haven't used Mox yet, but can we just provide a `ElixirALE.SPI.API` behaviour that the user can use for their mock? It seems like that would probably be the easiest for most people to use since it doesn't involve the port, and it seems reasonable for ElixirALE to provide the behaviour to encourage testing.

Thoughts?

adkron, 2018-03-08:
1. I agree that API changes should happen in 2.0 and not in 1.x. I think that would make it difficult for users to keep up, and I don't think it is necessary.

3. I think that using real hardware limits contributions to the project, and makes CI a pain. I like the idea of exercising the C as much as we can. That is an important part of the whole and could hide some errors. The issue that I have is that I don't know how well we can test the high level issues like the message interleaving. If we have a loopback in the C how do we inject responses and the orders in which they come in. We wouldn't have been able to find the `receive` bug if the response just cam straight back. I think we are in a position where we need the ability to test through the C, but also a higher level interface that bypasses the C and allows us a little more control.

4. I think that providing behaviors for each of the interfaces is a great idea that will help others test their libraries. They can then use Mox or whatever they like to fake the interface with ElixirALE. It also allows other people to implement a compatible version of ElixrALE which could lead to more choices, and more PRs coming in.

I still believe that we would be good to mock the calls to and from the ports so that we have more control in the tests.

fhunleth, 2018-11-02:
Closing since `elixir_ale` probably won't get any big updates going forward. Work is moving to https://github.com/elixir-circuits/.

closed 2018-11-02 by fhunleth