fhunleth/elixir_ale #57: Change direction of a GPIO pin.

amol-c, 2018-02-27:
Hi,

I am trying to interface with AM2302 sensor using [this datasheet](https://cdn-shop.adafruit.com/datasheets/Digital+humidity+and+temperature+sensor+AM2302.pdf). It requires the GPIO to be set to input direction for first few cycles and then set to output direction to start reading off the sensor.

I am trying to use the `set_int` in the input mode to listen for any change in the signal. But, I keep getting the below error

```
GPIO chip pinctrl-bcm2835: gpis_lock_as_irq: tried to flag a GPIO set as output for IRQ
```

It sounds like even after setting the output to input, the GPIO doesn't seem to revert to input.

Below is the code I am using.

```
spawn(fn-> temperature_sensor_input_state(23) end)

defp temperature_sensor_input_state(pin_number) do
    {:ok, input_pid} = GPIO.start_link(pin_number, :input)
    GPIO.write(input_pid, @low_value)
    sleep(1)

    GPIO.write(input_pid, @high_value)
    sleep(1)

    case GPIO.release(input_pid) do
       :ok -> 
         temperature_sensor_input_state(pin_number)
    end
  end

  defp temperature_sensor_input_state(pin_number) do
    {:ok, output_pid} = GPIO.start_link(pin_number, :output)

    returned_value = GPIO.set_int(output_pid, :both)
    listen()
  end

  defp listen() do
    receive do
      {:ok, sender, message} ->
       Logger.debug("Returned Value is sender = #{sender}, message = #{message}")
    end

    listen()
  end
```

Please advice.

Thanks.

fhunleth, 2018-02-27:
The protocol that you're looking for is call OneWire and ElixirALE doesn't support it. It is supported by the Raspberry Pi. A google search led me to this blog post which might help you: http://www.carstenblock.org/post/project-excelsius/. It looks good from a quick pass. You may also want to try other combinations of DHT-22 and Elixir and Nerves.

closed 2018-02-27 by fhunleth

amol-c, 2018-02-27:
Awesome ! Thanks !