fhunleth/elixir_ale #65: GPIO support active_low signals

mdinardo, 2018-12-11:
According to the sysfs gpio driver (https://www.kernel.org/doc/Documentation/gpio/sysfs.txt):
The device file "active_low" is available. Writing a "0" or a "1" to it makes the gpio normal or active_low, respectively.  Regarding rising and falling edge detection, those triggers will follow the active_low setting.

fhunleth, 2018-12-11:
Do you have a use case? Supporting active_low GPIOs seems like it would be confusing.

mdinardo, 2018-12-11:
In general, it is not uncommon to encounter hardware devices that employ Inverted (active_low) logic.  Ex: reset lines, chip selects, enable signals, etc.

As for me specifically, I have a large grouping of GPIO's.  On one of them is hooked up to an active-low enable signal of an IC that controls a power rail.  Two problems with this:
1.) In the time delay between finishing starting up the gpio device and manually writing a "1"/"true" to the gpio, the rail can blip on temporarily.  In the general use case, If this line is hooked up to something critical then bad things could happen.
2.) when orchestrating a large number of GPIO's, i now have to keep track of which ones are normal and which are inverted logic in order to choose the correct physical value.  it's so much easier to only have to consider the logical<->physical state conversion once during setup, and then for the rest of your code only work with its logical state (ie is it enabled or disabled?).  

regarding supporting this:  you just have to think about the gpio as its logical state instead of physical state, and understand that the OS will handle the conversion to the physical state.  Read-write code stays the same.

fhunleth, 2018-12-12:
Were you planning on sending a PR?

mdinardo, 2018-12-12:
yes, I just created it: https://github.com/fhunleth/elixir_ale/pull/66

fhunleth, 2018-12-30:
I've been thinking about this quite a bit, and I'm skeptical that logistic state is a helpful concept. For example, if I need to support someone who is verifying inputs and outputs against measurements by a logic analyzer or multimeter, I now need to check their logical state setting and explain how what they write to a GPIO may or may not match what they see. Where I've seen logical state used well is at the next layer up in the LED controllers, button handlers, etc. I'd like to keep a direct mapping between what you tell Elixir/ALE to do and what happens on the wire. I don't see that preventing you or anyone else from implementing a logical layer if you wanted.

As a practical matter, the PR is a decent change that hits on two codebases that are being superceded: 1. sysfs is being replaced by libgpiod in Linux and 2. Elixir/ALE is being replaced by circuits_gpio, circuits_i2c, and circuits_spi. I don't want to risk breaking the many Elixir/ALE users especially since much of the code using it is lightly unit tested at the hardware level.

The argument about glitching is a reason to make a change, but it affects setting any output to 1 on initialization (inverted or not inverted logic). I believe that can be fixed with libgpiod when we get to adding that to circuits_gpio. That will be important - thanks for bringing that up.

closed 2018-12-30 by fhunleth